dingo35/SmartEVSE-3.5 #244: Charging session started without starting OCPP session while OCPP enabled

malfunctioningEddy, 2025-06-29:
**Describe the bug**
- Car had been charging SOLAR 1P for some hours which was fine.
- I took the car and got home late (22:30) and connected charging cable because it needed charging. There was no sun so it didn't started charging.
- I needed 3P charging because I had limited time so I switched to SMART expecting to be meeted with "AUTHORIZE" on SmartEVSE as there was no open OCPP session
- Two things happened:
1 Charging started emmediatelly without me needing to scan my card. I checked that with my backend. Indeed no session was started. I fixed this by scanning the card during charging. OCPP session was started.
2 On the display it says "SMART 1P" while actually the car is charging with 3P

**Upload your config**
[settings.json](https://github.com/user-attachments/files/20970084/settings.json)

**Upload a debug log**
As this was the first instance I didn't start a debug session before charging. Will do that tomorrow.

**To Reproduce**
Steps to reproduce the behavior:
1. What Mode are you in? Normal, Smart or Solar?
Solar
2. What steps do you take so the problem arises?
From Solar to Smart

**Expected behavior**
A clear and concise description of what you expected to happen; expected Charging Current?
From Solar to Smart
Expecting to have to scan RFID
Charging at 3P

**Screenshots**
Only add screenshots if it adds any information, e.g. when you are reporting a problem on the webserver screen.
[image]

dingo35, 2025-06-30:
Which version is this? Did you compile it yourself, if so, from which repo/branch, if not, which download is it?

malfunctioningEddy, 2025-06-30:
Hi, it's [https://github.com/user-attachments/files/20881690/23b6c12a27583.zip](url) from issue 243.

dingo35, 2025-07-01:
Ok I can't reproduce this; the Smart1P on the display is a known bug, its only a display problem so you can ignore it.

But as soon as I connect an EV, it immediately says "Authorize"; I assume your RFIDreader is set at "Rmt/OCPP" , could you please check this, its not displayed in /settings ....

Now if I change modes by short pressing the left button, it shows "Ready" momentarily but immediately switches to "Authorize" again.

Can you reproduce this on your setup?

malfunctioningEddy, 2025-07-01:
Hi I tried yesterday again.
The only change in procedure is that I didn't take the car but just stoppen the Solar and started Smart.
Now I needed to Authorise so couldn't reporduce either.

Maybe it happended only after an amount of time. I will try to retest today but with all the sun maybe I won't need Smart mode now.

dingo35, 2025-07-01:
Appreciate your testing!

malfunctioningEddy, 2025-07-04:
Let's close this as you have released new version that could fix this.

closed 2025-08-09 by dingo35