dingo35/SmartEVSE-3.5 #247: One of my two chargers refuse to start charging with OCPP back-end (TAP ELECTRIC) enabled.

Adze1502, 2025-07-06:
**Describe the bug**

Summary: One of my two chargers refuse to start charging with OCPP back-end (TAP ELECTRIC) enabled.

Since a recent firmware update i have issues with my one of my EV chargers. I have two SmartEVSE chargers. Both use OCPP backend service from Tap Electric. Most of the time they worked fine until recently. Now one of my chargers refuses to start charging, no matter what i try. The other charger is functioning normally. I have tried so many things to resolve my issue and come to ask for your help.

What i see is that when a rfid tag is presented, the charger asks for authorisation. When authorisation is given and the start charge command is given the charger sends a reject message back. And i get an error in the backend logs. This is only with one of my chargers.

What i tried so far:
- Try start charging with RFID (fail)
- Try start charging from Tap Electric app (fail)
- Update to all kind of different version from 3.7.4 to 3.8.5 (fail)
- Create my own firmware using pio (fail)

I have tried the debug versions of different firmwares to see if any clues are shown on why starting fails, but cannot find anything of interest or any clue. I know from another tweaker ([sanderb2001](https://tweakers.net/gallery/548214) who has excatly the same issue, but he gave up on troubleshooting.

What does work:
- Local RFID scanning and authorisation. Learn RFID card, present card and charging starts.

What i still like to try but failed to do is a complete factory reset. Clear the complete configuration. I have a feeling that something is conflicting with the current setup. I did a full firmware flash with pio, but all my settings were retained. So this is not  a complete whipe. Would it help to do a complete whipe?

**Upload your config**
{"version":"v3.8.5","serialnr":45316,"mode":"OFF","mode_id":0,"car_connected":false,"wifi":{"status":"WL_CONNECTED","ssid":"XXXX","rssi":-62,"bssid":"18:E8:29:C7:8F:37"},"evse":{"temp":32,"temp_max":65,"connected":false,"access":0,"mode":1,"loadbl":0,"pwm":1024,"custombutton":false,"solar_stop_timer":0,"state":"Ready to Charge","state_id":0,"error":"None","error_id":0,"rfidreader":"Rmt/OCPP","rfid":"Ready to read card","rfid_lastread":"00000000000000"},"settings":{"charge_current":0,"override_current":0,"current_min":6,"current_max":20,"current_main":25,"current_max_circuit":20,"current_max_sum_mains":0,"max_sum_mains_time":0,"solar_max_import":0,"solar_start_current":4,"solar_stop_time":10,"enable_C2":"Always On","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0,"lcdlock":0,"lock":0,"cablelock":0},"mqtt":{"host":"192.168.201.10","port":1883,"topic_prefix":"SmartEVSE/7657","username":"smartevse-7657","password_set":true,"status":"Connected"},"ocpp":{"mode":"Enabled","backend_url":"wss://ocpp.tapelectric.app/xxxxxxxxxxxx","cb_id":"SmartEVSE-45316","auth_key":"xxxxxxxxxxxxxxxxxxxx","auto_auth":"Disabled","auto_auth_idtag":"XXXXXXXXXXXXXX","status":"Connected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"InvEastrn","address":240,"import_active_power":0,"total_kwh":1214.1,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":1214.1,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":49,"L1":19,"L2":14,"L3":16,"last_data_update":1751798796,"original_data":{"TOTAL":49,"L1":19,"L2":14,"L3":16}},"backlight":{"timer":0,"status":"OFF"},"color":{"off":{"R":0,"G":0,"B":0},"normal":{"R":0,"G":255,"B":0},"smart":{"R":0,"G":255,"B":0},"solar":{"R":255,"G":170,"B":0},"custom":{"R":0,"G":0,"B":255}}}

**Upload a debug log**
Nothing in the debug logs, can supply debug logs on request.

**To Reproduce**
1. Present RFID card
2. authorisation is given by OCPP backend
3. EV charger rejects start charging request: _{"idTag":"DD209641","connectorId":1}{"status":"Rejected"}_

**Expected behavior**
1. Present RFID card
2. authorisation is given by OCPP backend
3. EV charger starts charging

dingo35, 2025-07-06:
You can erase your settings by executing /erasesettings, but I wouldnt do that because I expect it to solve nothing.

You can produce the communication to the OCPP backend on your USB port  if you clone the v3.8.5 source code, and modify the MO_DBG_LEVEL to MO_DL_DEBUG, and connect to your USB port with baudrate 115200.

Preferable you would do this with a working version / working EVSE, and compare the communication with the faulty EVSE/firmware.

Another thing you could try is sending the (by the backend) expected RFID string in Auto-Authorize field, and see how that goes.

dingo35, 2025-07-06:
EDIT: perhaps superfluous: you modify the lines in platformio.ini, compile and upload.

I noticed a very strange serialnr, did you modify that manually?

I tried to reproduce but all works fine at my test setup.

Monitor your rfidlastread in /settings to see if all is recognized ok.

Please report back with you findings, but scramble your RFID string, since you dont want to pay for other peoples charging sessions.

Adze1502, 2025-07-07:
Here is the debug LOG:

[MO] debug (Connector.cpp:494): Status changed Available -> Preparing 
[MO] info (StatusNotification.cpp:52): New status: Preparing (connectorId 1)
[MO] Send: [2,"2cb5140c-5f42-5c04-d6c7-ac35948dfc64","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing","timestamp":"2025-07-07T15:50:53Z"}]
[MO] Recv: [3,"2cb5140c-5f42-5c04-d6c7-ac35948dfc64",{}]
[MO] Recv: [2,"64d40f7f-9878-4784-a0df-5e16a494b472","RemoteStartTransaction",{"idTag":"04XXXXXX901990","connectorId":1}]
[MO] debug (FilesystemAdapter.cpp:382): File open successful: /tx-1-69.jsn
[MO] debug (FilesystemUtils.cpp:73): Loaded JSON file: /tx-1-69.jsn
[MO] debug (TransactionDeserialize.cpp:269): DUMP TX (05XXXXXX810023)
[MO] debug (TransactionDeserialize.cpp:270): Session   | idTag 05XXXXXX810023, active: 0, authorized: 1, deauthorized: 0
[MO] debug (TransactionDeserialize.cpp:271): Start RPC | req: 1, conf: 1
[MO] debug (TransactionDeserialize.cpp:272): Stop  RPC | req: 1, conf: 1
[MO] debug (FilesystemAdapter.cpp:382): File open successful: /tx-1-66.jsn
[MO] debug (FilesystemUtils.cpp:73): Loaded JSON file: /tx-1-66.jsn
[MO] debug (TransactionDeserialize.cpp:269): DUMP TX (05XXXXXX810023)
[MO] debug (TransactionDeserialize.cpp:270): Session   | idTag 05XXXXXX810023, active: 0, authorized: 1, deauthorized: 0
[MO] debug (TransactionDeserialize.cpp:271): Start RPC | req: 1, conf: 1
[MO] debug (TransactionDeserialize.cpp:272): Stop  RPC | req: 1, conf: 0
[MO] debug (Connector.cpp:633): cannot delete more tx
[MO] ERROR (Connector.cpp:908): could not allocate Tx
[MO] Send: [3,"64d40f7f-9878-4784-a0df-5e16a494b472",{"status":"Rejected"}]

From what it looks is that the charger thinks that there is still a session active with tag 05XXXXXX810023. That's why it cant start a new session...

dingo35, 2025-07-08:
Ok I am not the author of the OCPP module, so we are going to do a small experiment here; I hope you are on Linux, and if not, you are able to translate this to your own OS yourself:
-pipx install esptool
-if you look in partitions_custom.csv, you see we have a partition called SPIFFS; don't be fooled, it is a LittleFS partition, but it is called SPIFFS for historical reasons; the partition starts at 0x370000 and has size 0x80000 .
-we are going to save this partition:
esptool read-flash 0x370000 0x80000 /tmp/save.bin
-now we are going to clear this partition:
esptool erase-region 0x370000 0x80000

-after restart you should notice your OCPP settings are gone; if you fill your settings you should be "starting from zero".

In case of problems you restore your save.bin:
esptool write-flash 0x370000 save.bin

Adze1502, 2025-07-08:
I will try! Get back to you soon. Just to be sure: on my partitions_custom.csv it says "spiffs,   data, spiffs, 0x370000,  0x90000," I should use 90000 instead of 80000 right?

Adze1502, 2025-07-08:
You my man, are the hero of the day! Did a "esptool erase-region 0x370000 0x90000" and now charger is working fine again!

Can i ask for a "secret" /erasespiffs feature request? Or is this just an edge case and not worth the effort? Thnx again, feel free to close this issue! 👍