dingo35/SmartEVSE-3.5 #249: MQTT message CableLock invalid

gerritjandebruin, 2025-07-13:
**Describe the bug**
The SmartEVSE announces to Home Assistant that the CableLock has options 0 and 1 (in `homeassistant/select/SmartEVSE-CableLock/config`) , but reports Disabled/ Enabled (in `SmartEVSE/CableLock`).

**Upload your config**
{
  "version": "v3.9.0",
  "serialnr": 7921,
  "mode": "NORMAL",
  "mode_id": 1,
  "car_connected": true,
  "wifi": {
    "status": "WL_CONNECTED",
    "ssid": "de Bruin IoT",
    "rssi": -70,
    "bssid": "EE:15:A2:45:A8:13"
  },
  "evse": {
    "temp": 37,
    "temp_max": 65,
    "connected": true,
    "access": 1,
    "mode": 0,
    "loadbl": 0,
    "pwm": 102,
    "custombutton": false,
    "solar_stop_timer": 0,
    "state": "Connected to EV",
    "state_id": 1,
    "error": "None",
    "error_id": 0,
    "rfidreader": "Disabled",
    "rfid": "Not Installed"
  },
  "settings": {
    "charge_current": 60,
    "override_current": 0,
    "current_min": 6,
    "current_max": 16,
    "current_main": 35,
    "current_max_circuit": 16,
    "current_max_sum_mains": 0,
    "max_sum_mains_time": 0,
    "solar_max_import": 0,
    "solar_start_current": 5,
    "solar_stop_time": 10,
    "enable_C2": "Not present",
    "mains_meter": "API",
    "starttime": 0,
    "stoptime": 0,
    "repeat": 0,
    "lcdlock": 0,
    "lock": 1,
    "cablelock": 0
  },
  "mqtt": {
    "host": "192.168.1.240",
    "port": 1883,
    "topic_prefix": "SmartEVSE",
    "username": "bruingjde",
    "password_set": true,
    "status": "Connected"
  },
  "ocpp": {
    "mode": "Disabled",
    "backend_url": "",
    "cb_id": "",
    "auth_key": "",
    "auto_auth": "Disabled",
    "auto_auth_idtag": "",
    "status": "Disconnected"
  },
  "home_battery": {
    "current": 0,
    "last_update": 0
  },
  "ev_meter": {
    "description": "API",
    "address": 10,
    "import_active_power": 0,
    "total_kwh": 241.3,
    "charged_kwh": 0,
    "currents": {
      "TOTAL": 0,
      "L1": 0,
      "L2": 0,
      "L3": 0
    },
    "import_active_energy": 241.3,
    "export_active_energy": 0
  },
  "mains_meter": {
    "import_active_energy": 0,
    "export_active_energy": 0
  },
  "phase_currents": {
    "TOTAL": 10,
    "L1": 10,
    "L2": 0,
    "L3": 0,
    "last_data_update": 1752390209,
    "original_data": {
      "TOTAL": 10,
      "L1": 10,
      "L2": 0,
      "L3": 0
    }
  },
  "backlight": {
    "timer": 0,
    "status": "OFF"
  },
  "color": {
    "off": {
      "R": 0,
      "G": 0,
      "B": 0
    },
    "normal": {
      "R": 0,
      "G": 255,
      "B": 0
    },
    "smart": {
      "R": 0,
      "G": 255,
      "B": 0
    },
    "solar": {
      "R": 255,
      "G": 170,
      "B": 0
    },
    "custom": {
      "R": 0,
      "G": 0,
      "B": 255
    }
  }
}

**Upload a debug log**
(hopefully not needed)

**To Reproduce**
Problem was present in earlier releases as well. Still present after update to 3.9.0.

**Expected behavior**
I think 
https://github.com/dingo35/SmartEVSE-3.5/blob/48133874c39ade5a06ec876ec0026e723b6a5487/SmartEVSE-3/src/esp32.cpp#L922
should report just 0 or 1 (or the announcement of MQTT configuration should report options ["Enabled", "Disabled"] in https://github.com/dingo35/SmartEVSE-3.5/blob/48133874c39ade5a06ec876ec0026e723b6a5487/SmartEVSE-3/src/esp32.cpp#L852

**Screenshots**

[image]

dingo35, 2025-07-13:
Thanks for your report, you are absolutely right. 

Although I would prefer the values to be 0 and 1 (seems cleaner to me), I think for backwards compatibility reasons it is better to keep the "Enabled"/"Disabled" values, and announce them accordingly.

Could you please test if this version behaves ok for HA?

[0c262817.zip](https://github.com/user-attachments/files/21203078/0c262817.zip)

gerritjandebruin, 2025-07-13:
Thanks for the quick reply!

It does unfortunately not work correctly yet. Altough now for HA this is all right, I think the publishing by HA to topic `/Set/CableLock` does not work correctly. It will now publish either `Enable` or `Disable` to that topic, but that does not change the setting (as observed in the WebUI and by reading `CableLock` topic). Publishing manually `0` or `1` to `/Set/CableLock` does change the setting.

Maybe something needs to be changed at https://github.com/dingo35/SmartEVSE-3.5/blob/48133874c39ade5a06ec876ec0026e723b6a5487/SmartEVSE-3/src/esp32.cpp#L661 as well?

dingo35, 2025-07-14:
Ok in that case we go back to the old announcement, everything with 0 and 1 now, no Enabled/Disabled; this is better in line with the REST API also .

Please test:

[1aa1f30ac72.zip](https://github.com/user-attachments/files/21210533/1aa1f30ac72.zip)

gerritjandebruin, 2025-07-14:
Yes, that one works!

dingo35, 2025-07-14:
Pushed in d924e9e8b80122d3c7553606fc0d89eaea59110c

closed 2025-07-14 by dingo35