dingo35/SmartEVSE-3.5 #254: Interested in EVCC integration?

maximevince, 2025-07-22:
**Is your feature request related to a problem? Please describe.**
I'd like to integrate Smart EVSE to EVCC (https://evcc.io/en/)
I've currently done this partially through the REST API, but it's not ideal, a bit slow and polling based.
A proper MQTT integration would be cleaner.
EVCC requires exposing the current J1772 status (A,B,C,D,...) and some other values in a specific format. (which are currently not exposed or in a different format)
I've got a work-in-progress branch with most of this implemented.
Are you guys interested in something like this?

**Describe the solution you'd like**
If the SmartEVSE maintainers have any interest in this, I could create a PR.

**Describe alternatives you've considered**
I could not send a PR :)

**Additional context**
If yes, we can discuss your preferences on the names of the MQTT topics etc... (I was thinking about putting them under a separate /evcc/ tag, with exactly the names of the variables that the EVCC yaml config requires.

dingo35, 2025-07-22:
Wouldnt this just be a matter of translating MQTT topics and values?

I consider that functionality that should be done by the MQTT broker.

I'd like to guard function creep because with upcoming functionality we are over 99% of Flash Ram use.... which makes it pretty scarce.

If you need any new topics subscribed or publised, I'd consider that in a PR (since it might apply to other users/applications).

maximevince, 2025-07-23:
> Wouldnt this just be a matter of translating MQTT topics and values?
Not really, some information is just not exposed now.

>I'd like to guard function creep because with upcoming functionality we are over 99% of Flash Ram use.... which makes it pretty scarce.
That totally makes sense. I'll look into only exposing the J1772 status, which might be enough.

I'll keep you posted

Timple, 2025-07-23:
Curious bystander here. Would this conflict with the EVSE solar or smart mode?

maximevince, 2025-07-24:
@Timple yeah, it would conflict with the SmartEVSE solar mode.
I've currently set-up EVCC to switch between `NORMAL` and `OFF`, where EVCC can then set the max current in normal mode, to match what EVCC wants (e.g. solar surplus in evcc solar mode).

maximevince, 2025-07-24:
There might be another problem, too:
SmartEVSE doesn't update the J1772 status when in `OFF` mode (understandably so), and it'll always be `A`.
But EVCC will only start charging when J1772 state is B (or C).
Maybe this can be solved by using "PAUSE" instead - let's see.

maximevince, 2025-07-24:
So, as far as I can tell "PAUSE" works for this, partially:
Whenever the EV is connected, and "PAUSE" is entered, it will go to "B1" mode. This allows EVCC to see the EV as still connected, and it can be resumed when EVCC decides to continue charging.
However, when the EV gets disconnected - while SmartEVSE is in PAUSE mode, it will go to state A, and stay there. It won't try to detect that a new EV is connected and therefor never go to state B. This causes EVCC to be unable to start changing the vehicle anymore, since it thinks there's no EV connected.

Question for SmartEVSE maintainers: I suppose this is intended behavior. Would it be allowed to detect an EV being connected in PAUSE mode, and therefore going from A -> B state? It would never go to C state, while in PAUSE mode. To me, that sounds like an improved PAUSE mode behavior, but I also know you guys have a lot of other considerations - such as OCCP integration. So please educate me if this is possible or not.