dingo35/SmartEVSE-3.5 #258: Blue led in Solar mode instead of yellow?

rhannink, 2025-08-05:
Since the latest version 3.9.1 the led on my charger is blue instead of yellow in solar mode.
Is this expected behaviour?

Gr. Remco

dingo35, 2025-08-06:
I cannot reproduce that, perhaps you did something with custombuttons?
If not post your /settings ...

rhannink, 2025-08-08:
Here is my raw config:

`{"version":"v3.9.1","serialnr":6828,"mode":"SOLAR","mode_id":2,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"hannink.org","rssi":-63,"bssid":"F6:9F:C2:F7:C5:23"},"evse":{"temp":44,"temp_max":65,"connected":true,"access":1,"mode":2,"loadbl":0,"pwm":102,"custombutton":false,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfidreader":"Disabled","nrofphases":3,"rfid":"Not Installed"},"settings":{"charge_current":60,"override_current":0,"current_min":6,"current_max":16,"current_main":25,"current_max_circuit":16,"current_max_sum_mains":75,"max_sum_mains_time":0,"solar_max_import":3,"solar_start_current":3,"solar_stop_time":10,"enable_C2":"Auto","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0,"lcdlock":0,"lock":0,"cablelock":0},"mqtt":{"host":"127.0.0.1","port":1883,"topic_prefix":"SmartEVSE-6828","username":"","password_set":false,"status":"Disconnected"},"ocpp":{"mode":"Disabled","backend_url":"","cb_id":"","auth_key":"","auto_auth":"Disabled","auto_auth_idtag":"","status":"Disconnected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"InvEastrn","address":12,"import_active_power":4,"total_kwh":7623.4,"charged_kwh":6.2,"currents":{"TOTAL":176,"L1":58,"L2":58,"L3":60},"import_active_energy":7623.4,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":140,"L1":50,"L2":40,"L3":50,"last_data_update":1754649438,"original_data":{"TOTAL":140,"L1":50,"L2":40,"L3":50}},"backlight":{"timer":120,"status":"ON"},"color":{"off":{"R":0,"G":0,"B":0},"normal":{"R":0,"G":255,"B":0},"smart":{"R":0,"G":255,"B":0},"solar":{"R":255,"G":170,"B":0},"custom":{"R":0,"G":0,"B":255}}}`

I did nothing with the custom colors in the config. I do not even know how to change these. The Button now is blue when the charger is switched to off. The button was alway not lit when the charger was off.

dingo35, 2025-08-08:
On my test bench, when I take over your settings, and switching from Solar to Off the LED in the switch goes off.

You are using the switch with built in LED from stegen.com ?

rhannink, 2025-08-11:
If I revert to v3.8.5 all the colors of the switch are ok again. Light off when switched off an yellow with solar mode.

I then tried 3.9.1 again and the problem is back. Blue ring when switched off and sometimes blue pulsing when waiting for solar???

rhannink, 2025-08-13:
I need some help with git I am afraid.

I can compile the source code and that works.

```
git clone https://github.com/dingo35/SmartEVSE-3.5.git
cd SmartEVSE-3/SmartEVSE-3
platformio run.
```

To choose a specific commit is use
```
cd ..
git fetch origin 48133874c39ade5a06ec876ec0026e723b6a5487
cd SmartEVSE-3
platformio run
```
Is this the correct way? I feel that I am alwas running the latest version/commit HEAD instead of the specific commit version.

rhannink, 2025-08-15:
Found it,

Commit 84b0888 is still working Ok.
Commit bd9c4b1 is showing the blue flashing ring.

When I look at the code (I am no programmer in any way..) I see that it is using the custom color and my custom color is also set to blue. I don't no exactly where it goes wrong, but it is somewhere in commit bd9c4b1.

Hope this helps.

Gr. Remco

rhannink, 2025-08-20:
The zip file contains only a firmware.elf file. Can I just rename this to firmware.bin ?

Imaginous, 2025-08-20:
NO.... my bad... New file coming up

PR #267 (2025-08-20): Fixed bug with wrong LED colors
Fixes #258 

This fixes the wrong LED colors since the addition of always publishing the button state.

I introduced a  seperate variable for the MQTT button state.
Re-using the CustomButton variable was a bad idea since it was also used for the LED state.