dingo35/SmartEVSE-3.5 #260: Feature request: switch back earlier when EnableC2=AUTO and NrOfPhases = 3 in Solar mode

dingo35, 2025-08-06:
See https://gathering.tweakers.net/forum/list_message/83103586#83103586

dingo35, 2025-08-06:
Please test:

[59d39db60cd.zip](https://github.com/user-attachments/files/21619935/59d39db60cd.zip)

gastje01, 2025-08-12:
Doesn't work as expected. When there is enough power to switch to 3P, charging stops, and starts again on 1P 6A while there is -19A available on mains. 

The C2 contactor never gets engaged.

gastje01, 2025-08-24:
Added some solar and the last few days were perfect for testing (sunny with a few clouds). Switching back goes much faster and this way I'm able to charge mostly on PV-energy only, perfect for when I don't need the car for a few days. 

I'm really happy with the way this works now. Charging on PV only, and only do a "top-up" during the night (Smart mode) when I need more charge after the sun went down. 

Would you like me to test something specific?

dingo35, 2025-08-25:
Ok thx for testing, I'll take this to the next release.

closed 2025-08-25 by dingo35

dingo35, 2025-08-25:
Have you tested the version above?