dingo35/SmartEVSE-3.5 #262: v3.1 hardware problems with Eastron modbus

dingo35, 2025-08-09:
Two separate users have problems getting the communication with Eastron 3 phase meters to work.

See  https://gathering.tweakers.net/forum/list_message/83124728#83124728

dingo35, 2025-08-09:
My meter is type SDM72D-M shows 03 01.03  .

MullieC, 2025-08-09:
After a reboot of the EVSE this error occur shows up:
[image]

After a while te error disappears and the EVSE will continue to work:
[image]

MullieC, 2025-08-09:
This is the software version of the Eastron SDM72D-M kWh meter:
[image]

This is the serial number of the Eastron SDM72D-M kWh meter:
[image]

This is the Modbus address of the Eastron SDM72D-M kWh meter:
[image]

This is the Baudrate setting of the Eastron SDM72D-M kWh meter:
[image]

This is the Parity setting of the Eastron SDM72D-M kWh meter:
[image]

Just to be sure the problem is not caused by this kWh meter i have also tried 2 others of the same type and software version with the same settings:
[image]

jaimevisser, 2025-08-09:
(Tassadar32 on Tweakers.net)

Tried connecting to my Eastron meter using an DSD tech SH-U10 and can get correct readings. I was lazy so I had claude (AI) read as many registers as possible using mbpoll. Results below:

[image]

Commands used look like this:
`mbpoll -m rtu -a 10 -b 9600 -P none -t 3:float -r 49 -c 30 -1 -B /dev/cu.usbserial-0001`

Some pictures of my meter:

[image]

[image]

[image]

[image]

[image]

[image]

jaimevisser, 2025-08-10:
Even if C2 is set to always off? Does the EVSE firmware check readings on phases that will never be used? Sounds like a bug to me. I only had L1 connected but will patch it through to L2&3 for testing. I'll see if this makes a difference.

jaimevisser, 2025-08-10:
Oh, that's weird. Didn't do that and was still able to read the values using modbus dongle and see kWh go up on one phase charging.

dingo35, 2025-08-10:
It depends on the model: one SDM72-DMv2 behaves differently from the next, for some reason they hardly increase the version number....

jaimevisser, 2025-08-10:
Nope, connecting power to all three phases doesn't solve the problem.

MullieC, 2025-08-10:
@dingo35 
After boot, the error occurs after about 2 minutes. The SmartEVSE shows an error on its display and stops charging the car. After about 5 minutes, the error message on the display disappears, and the SmartEVSE becomes fully operational, allowing the car to start charging again. After a couple of minutes, the error occurs again and the SmartEVSE stops working. This loop of working and error will continue.
See the list below I have obtained from error mode in Telnet and stopwatch:
Time	event
0:00    Reboot of the SmartEVSE
0:55    CRC check error
1:16    CRC check error
1:52    CRC check error
1:55    CRC check error
1:58    CRC check error
2:00    SmartEVSE shows error in the display and stops working. From now on about every 2 seconds the CRC check error shows up.
6:40    SmartEVSE has no more error in the display and starts working normal 
6:40    CRC check error
6:50    CRC check error
7:10    CRC check error
7:15    CRC check error
7:55    CRC check error
8:10    CRC check error

I have also tried multiple adresses (1, 2, 10, 11, 12, 20, 32, 100, 101, 123, 200, 201, 234). There is communication between the SmartEVSE and the Easton kWh meter: the kWh meter shows the phone symbol on the display as long as the cable is connected and the EVSE and kWh meter do not have the same address. There are also current values visible on the EVSE that match the kWh meter values.

Yes, my test setup is single-phase. I have also connected the kWh meter with only one phase and neutral. However, for testing purposes, I connected the other two phases of the kWh meter to the same phase as well. This made no difference. I also want to use the solar mode, and in that case, there is a chance that charging will only take place on one phase if the other two phases are switched off due to insufficient power from the solar panels.

There is no difference if I disable the mains meter and set the EV meter to address 32. The fault persists. I even see the errors below appear in Normal mode while woring properly when charging the car if I leave one of the two meters enabled. Only when I disable both meters in the Normal mode there is no longer any fault in Telnet.
The (MBhandleError)(C1) Error response: E2 - CRC check error, address: 20, function: 04, reg: 0048
The (MBhandleError)(C1) Error response: E2 - CRC check error, address: 20, function: 04, reg: 0006

jaimevisser, 2025-08-10:
Mine is still mostly a portable testbench. If it helps and you are located anywhere near either Oss or Utrecht (or anyplace inbetween) I can just let you borrow mine for testing.

mstegen, 2025-08-22:
I've done some tests to see what causes this issue. And find a way to fix it.
The issue is that while receiving RS485 data, the eModbus library reads an extra byte, this byte can be 0x00 or 0xf0, causing the CRC error.
The cause of this is the switching between TX/RX by the transceiver chip. This causes a tiny glitch on the RX line, which is read as a data byte by some ESP32 chips.
The really weird thing is that most of the controllers are fine, and don't have this issue.

On the affected controllers adding a 10k pull-up resistor on the RS485_RX line (pin1) of the transceiver chip fixes the error.
If you have one of these controllers with this issue, please contact me by e-mail (michael@stegen.com) and i'll exchange them for a fixed/new one.

[image]

mstegen, 2025-08-24:
@jaimevisser No, if you can do it yourself, please do so. It will not void the warranty, as long as you take ESD precautions while soldering to the circuit board.

@MullieC Only some of the new SmartEVSE v3.1 are affected.
On this hardware i used the ESP32-MINI-1, as you can see on your photo, instead of the normal ESP32. This chip has two extra I/O lines, but uses different I/O pins. So at startup a check is made to see what hardware is used, and the correct pins are then initialised.
For the RS485_RX line, GPIO37 is used, which is a input only pin.

jaimevisser, 2025-08-26:
Added the resistor, communication works! I can see both kWh and kW on the display during charging and can confirm the usage also shows up in the sessions tab in tapElectric.

lasharor, 2026-02-25:
How can I confirm if I have the same problem? I commented also on the tweakers.net topic. I get the following errors:

`(D) (ModbusRequestLoop)(C1) ModbusRequest 4: Request Energy Node 0
(V) (ModbusSend8)(C1) Sent packet address: 0x66, function: 0x04, reg: 0x0048, token:0x66040048, data: 0x0002.
[0;32m(V) (ModbusSend8)(C1) Sent packet address: 0x66, function: 0x04, reg: 0x0048, token:0x66040048, data: 0x0002.
(MBhandleError)(C1) Error response: E2 - CRC check error, address: 66, function: 04, reg: 0048.
(D) (ModbusRequestLoop)(C1) ModbusRequest 20: Request EVMeter Current Measurement Node 0
(V) (ModbusSend8)(C1) Sent packet address: 0x66, function: 0x04, reg: 0x0006, token:0x66040006, data: 0x000c.`

I bought the EVSE at the end of last year. With a 1 phase Eastron meter it will sometimes work for up to 1 hour. Most often it is way less (15 to 30 minutes). tried with mutiple cables.

mstegen, 2026-02-27:
If your SmartEVSE gives communication errors, every now and then, then it's probably related to this issue. I test every controller before i ship them out, and usually it's immediately clear that the controller has this issue when i do a communication test.
It looks like this one slipped through, please send me an e-mail (michael@stegen.com), and i'll swap the controller for another one.

jaimevisser, 2026-02-27:
Issue is back on 3.10.2, had to downgrade back to 3.9.2/291b again to fix.

harmenbrauns, 2026-02-27:
I think i might running in the same issue.

Debug info gives this crc error, don't know if this matters but I'm running the latest fw 3.10.2, never looked at the debug before, but what I see from HA is regular com errors.
Below the filtered debug info on errors only. Would this SmartEVSE need the resistor mod?

```
:321055ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
t:321151ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.
t:323280ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.
t:325347ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
t:325445ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.
t:327442ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
t:329649ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.
t:331800ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.
t:333785ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
t:333883ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.
t:335877ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
t:335972ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.
:335877ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
t:335972ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.
t:340106ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
:340106ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
t:340206ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.
t:342256ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
t:342355ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.
t:344369ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
t:346465ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
t:346564ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.
t:348562ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
t:348660ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.
t:350657ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
t:350755ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.
t:354965ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.
t:356991ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
t:359149ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
t:359247ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.
t:361262ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
t:363479ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.
t:365487ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
t:367596ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
t:367693ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.
:367596ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
t:367693ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.
t:371796ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
t:371894ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.
t:373884ms) (MBhandleError)(C1) Error response: E5 - Packet length error, address: 0c, function: 04, reg: 0048.
t:373988ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.
t:376023ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
t:378168ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
t:378268ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.
t:380326ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
t:380425ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.
t:382550ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.
[... cut ...]
t:388951ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.
t:390957ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
t:391058ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.
t:397374ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0048.
t:397470ms) (MBhandleError)(C1) Error response: E2 - CRC check error, address: 0c, function: 04, reg: 0006.

```

harmenbrauns, 2026-02-27:
> Are you using the Eastron meter as MainsMeter or EVmeter? or both?

Just as EVmeter. 

`{"version":"v3.10.2","serialnr":8330,"mode":"OFF","mode_id":0,"car_connected":false,"wifi":{"status":"WL_CONNECTED","ssid":"ap-brauns","rssi":-73,"bssid":"18:FD:74:C1:95:A2"},"evse":{"temp":21,"temp_max":65,"connected":false,"access":0,"mode":1,"loadbl":0,"pwm":1024,"custombutton":false,"solar_stop_timer":0,"state":"Ready to Charge","state_id":0,"error":"None","error_id":0,"rfidreader":"Disabled","nrofphases":1,"rfid":"Not Installed"},"settings":{"charge_current":0,"override_current":0,"current_min":6,"current_max":13,"current_main":25,"current_max_circuit":16,"current_max_sum_mains":75,"max_sum_mains_time":0,"solar_max_import":0,"solar_start_current":1,"solar_stop_time":10,"enable_C2":"Always Off","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0,"lcdlock":0,"lock":1,"cablelock":0},"mqtt":{"host":"192.168.73.2","port":1883,"topic_prefix":"SmartEVSE/8330","username":"","password_set":false,"tls":false,"status":"Connected","smartevse_server":false},"ocpp":{"mode":"Disabled","backend_url":"","cb_id":"","auth_key":"","auto_auth":"Disabled","auto_auth_idtag":"","status":"Disconnected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Eastron3P","address":12,"import_active_power":0,"total_wh":4284355,"charged_wh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":4284355,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":9,"L1":6,"L2":11,"L3":-8,"last_data_update":1772188254,"original_data":{"TOTAL":9,"L1":6,"L2":11,"L3":-8}},"backlight":{"timer":0,"status":"OFF"},"color":{"off":{"R":0,"G":0,"B":0},"normal":{"R":0,"G":10,"B":0},"smart":{"R":0,"G":0,"B":10},"solar":{"R":10,"G":10,"B":0},"custom":{"R":0,"G":0,"B":0}}}`

Alwik, 2026-08-27:
Hi all,
I have exactly the same problem, but with hardware V3. Could the solution be the same?
Best, Alwik

mstegen, 2026-08-27:
I have not heard any problems with v3 hardware.
v3 hardware used different pin on the ESP32, and used a WROOM chip, not the Mini as used in the v3.1

Imaginous, 2026-08-27:
I installed an Eastron SDM120-M yesterday on a V3 usimg address 12 without any problems.

Alwik, 2026-08-28:
Ok, thanks for the replies. I have the V3 hardware + Eastron SDM230 1-phase. This used to work fine (for a year or so), but then started to give more and more problems: exactly the one described earlier: quick red flashing, and an error complaining about communication with the ev-meter. I have replaced the Eastron and the cables (twice), but the problem remains. Is there anything I can do to figure out what could be the problem?

dingo35, 2026-08-28:
So something in your neighbourhood has started to disturb your modbus communication.
-use twisted pair cabling only
-terminate the ends with 120 ohm resistors
-experiment with connecting/disconnecting the ground on the A/B/ground connectors

If that all doesnt work you could try setting your Eastron to a high address like 112 decimal, for some this has worked better.

Alwik, 2026-08-28:
Thanks for your list of possible solutions @dingo35 !
The charge point is in a small box outside, there is nothing electrical in a radius of 5 meters around it.
I'm using twisted-pair, the distance is only 10 cm's or so.
The Eastron is already set to address 101(65).
I have added two 120 ohm resistors, will let you know the results!

Alwik, 2026-08-28:
Ok, I honestly didn't expect the resistors to work, but my charge point has had no EV meter issues for the past 7 hours, so adding the 2 resistors seems to have been the solution. Bedankt @dingo35 !

Imaginous, 2026-08-28:
Officially RS485 needs 120ohm termination resistors.

But it also signals there might be a strong interference source.

Alwik, 2026-08-29:
Unfortunately it didn't last: This morning again: 'EV Meter Comm Error'.
So I thinks I'll try adding a ground-wire between EVSE and kWh-meter, hope that helps...

Imaginous, 2026-08-29:
I don't know the schematics of the SmartEVSE, but it's possible (without a ground wire) the voltage drifted away to a high current compared to ground.

The delta between A/B is still fine, but the towards ground it can be hundreds of volts. This can become out of spec of the used RS485 chip.

This may lead to errors and potentially damage the RS485 chip on one of the connected bus devices.

Alwik, 2026-09-01:
I waited a couple of days before giving results this time, to be sure it worked, but:
Adding the ground-wire seemed to have done the trick! No more EV-Meter problems any more. Thanks for everybody's help!

Alwik, 2026-09-08:
Unfortunately the problem came back after a week. This time I had the debug-firmware running. A telnet gave me this:
(D) (ModbusRequestLoop)(C0) ModbusRequest 4: Request Energy Node 0
(V) (ModbusSend8)(C1) Sent packet address: 0x65, function: 0x04, reg: 0x0048, token:0x65040048, data: 0x0002.
(MBhandleError)(C1) Error response: E0 - Timeout, address: 65, function: 04, reg: 0048.
(D) (ModbusRequestLoop)(C1) ModbusRequest 20: Request EVMeter Current Measurement Node 0
(V) (ModbusSend8)(C1) Sent packet address: 0x65, function: 0x04, reg: 0x0006, token:0x65040006, data: 0x000c.
(MBhandleError)(C1) Error response: E0 - Timeout, address: 65, function: 04, reg: 0006.

Meter is working fine (display is working), cables changed, termination resistors there, ground connected...
The problem was there today for hours, but after a reboot of the SmartEVSE everything worked again immediately.
Current state is that it works intermittently. If it works, this is the log:
(D) (ModbusRequestLoop)(C0) ModbusRequest 4: Request Energy Node 0
[1;32m(D) (ModbusRequestLoop)(C0) ModbusRequest 4: Request Energy Node 0
(V) (ModbusSend8)(C1) Sent packet address: 0x65, function: 0x04, reg: 0x0048, token:0x65040048, data: 0x0002.
(V) (ModbusDecode)(C1) Received packet (7 bytes) 65 04 04 45 b7 93 a4
(V) (ModbusDecode)(C1)  valid Modbus packet: Address 0x65 Function 0x04 Register 0x0048 Response
(D) (ModbusRequestLoop)(C1) ModbusRequest 20: Request EVMeter Current Measurement Node 0
(V) (ModbusSend8)(C1) Sent packet address: 0x65, function: 0x04, reg: 0x0006, token:0x65040006, data: 0x000c.
(V) (ModbusDecode)(C1) Received packet (27 bytes) 65 04 18 41 76 0c c3 00 00 00 00 00 00 00 00 45 60 12 0c 00 00 00 00 00 00 00 00
(V) (ModbusDecode)(C1)  valid Modbus packet: Address 0x65 Function 0x04 Register 0x0006 Response

I am tempted to just order a new SmartEVSE at the moment, because I have replaced everything else...
Or is there a last thing I can check?

dingo35, 2026-09-08:
It is probably some electronic source that disturbs your modbus; might be picked up by the 230V cable. 
You can try buying a new SmartEVSE, I think its a small chance but you never know until you fixed the problem...

Remco55, 2026-09-10:
Yesterday I had the same problem with the Smart EVSE version 3.0, Sensorbox and Eastron SDM72D-M, all bought late 2023.
I had not connected the RS485 wires from the Eastron to the EVSE until yesterday.
After re-wiring, the communication error occurred.
However, when reverting back to the non-RS485 situation, the configuration error remained.
After a lot of trial and error, the following solution worked.

1. All AC electrical power off using the Breaker in the meter cabinet.
2. Pull both the P1 cable and the green connector from the Sensorbox.
3. Re-wire to the desired configuration.
4. Engage the breaker in the meter cabinet to restore AC power.
5. Connect the P1 cable and the green connector to the sensorbox.
6. Now the communication error is gone and everything works as advertised.

Ecno92, 2026-09-11:
I am also experiencing this issue on a SDM630 modubs.
I noticed it when I came back from a holiday and the other car was connected for >10 days. 

```
(I) (loop)(C1) STATE: A Error: 10 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A, MainsMeter.Timeout=0, CircuitMeter.Timeout=20, EVMeter.Timeout=0.
(MBhandleError)(C1) Error response: E0 - Timeout, address: 0a, function: 04, reg: 0000.
```

After resetting the device everything worked fine for more than a week.

So far I am not sure yet how to reproduce the issue, so that I can figure out what the root cause is.

In the logs you can see that EVMeter.Timeout=0. This seems to happen in case no successful message has come in for some time. 
Interestingly also the MainsMeter is at 0. So this specific communication error is not limited to the Eastron device.