dingo35/SmartEVSE-3.5 #264: ecxeed max main in normal mode

BarryH120, 2025-08-10:
Apologies upfront. Not sure if this is a feature of a bug.

**Describe the bug**
When selecting Normal mode the charger start charging with 32A (and my car uses 1A less), I would expect the charger to start with 25A (and stay with that)
setup:
1phase charging in a 3ph household. 25A fuse in meterboard, after the 25A fuse a 32A fuse for the smartevse and a 16A fuse with a solar inverter.
When selecting Smart mode the charger starts at 25A.

Question: 
0/ Is my description clear
1/ Is this expected behaviour (and in my situation I should never use Normal mode)?

**Upload your config**
{"version":"v3.8.5","serialnr":7416,"mode":"NORMAL","mode_id":1,"car_connected":true,"wifi":{"status":"WL_CONNECTED","ssid":"B2","rssi":-66,"bssid":"F8:0D:A9:09:B9:19"},"evse":{"temp":34,"temp_max":65,"connected":true,"access":1,"mode":0,"loadbl":0,"pwm":545,"custombutton":false,"solar_stop_timer":0,"state":"Charging","state_id":2,"error":"None","error_id":0,"rfidreader":"Disabled","rfid":"Not Installed"},"settings":{"charge_current":320,"override_current":0,"current_min":6,"current_max":32,"current_main":25,"current_max_circuit":16,"current_max_sum_mains":59,"max_sum_mains_time":0,"solar_max_import":0,"solar_start_current":6,"solar_stop_time":5,"enable_C2":"Always Off","mains_meter":"API","starttime":0,"stoptime":0,"repeat":0,"lcdlock":0,"lock":0,"cablelock":0},"mqtt":{"host":"","port":1883,"topic_prefix":"SmartEVSE-7416","username":"","password_set":false,"status":"Disconnected"},"ocpp":{"mode":"Disabled","backend_url":"","cb_id":"","auth_key":"","auto_auth":"Disabled","auto_auth_idtag":"","status":"Disconnected"},"home_battery":{"current":0,"last_update":0},"ev_meter":{"description":"Disabled","address":12,"import_active_power":0,"total_kwh":0,"charged_kwh":0,"currents":{"TOTAL":0,"L1":0,"L2":0,"L3":0},"import_active_energy":0,"export_active_energy":0},"mains_meter":{"import_active_energy":0,"export_active_energy":0},"phase_currents":{"TOTAL":320,"L1":10,"L2":0,"L3":310,"last_data_update":1754858842,"original_data":{"TOTAL":320,"L1":10,"L2":0,"L3":310}},"backlight":{"timer":119,"status":"ON"},"color":{"off":{"R":0,"G":0,"B":0},"normal":{"R":0,"G":255,"B":0},"smart":{"R":0,"G":255,"B":0},"solar":{"R":255,"G":170,"B":0},"custom":{"R":0,"G":0,"B":255}}}

**Upload a debug log**
Flash the debug version of the firmware (see https://github.com/dingo35/SmartEVSE-3.5/blob/master/SmartEVSE-3/HowToFlash.txt), telnet to your device, capture the debug log, cut it so that it shows JUST BEFORE and JUST AFTER the problem arises, and upload it here.

**To Reproduce**
Steps to reproduce the behavior:
1. What Mode are you in? Normal, Smart or Solar? Normal
2. What steps do you take so the problem arises? asap to Smart

**Expected behavior**
25A charging (in Normal mode)

**Screenshots**
Only add screenshots if it adds any information, e.g. when you are reporting a problem on the webserver screen.

dingo35, 2025-08-11:
In Normal mode the SmartEVSE will start charging at MaxCurrent, and it will disregard any settings for MaxMains and MaxCircuit; thats why you dont see any settings for those parameters in Normal mode.

You should change MaxCurrent to 25A or keep your current settings and select Smart mode.

Ususally, if you have a MainsMeter configured, you would never use Normal mode.

dingo35, 2025-08-11:
"Limit charging in Normal mode to the max of MaxMains, MaxCircuit and MaxCurrent , for those values that are defined."  

That is called Smart mode; it wouldnt make sense to define two equal modes.

Just set it in Smart mode, and check the LCD lock box on your dashboard, and other members of your household will no longer be able to screw up your settings.