dingo35/SmartEVSE-3.5 #272: Controller does not receive Homewizard P1 data.

joyvanacker, 2025-08-31:
**Describe the bug**
Smartevse does not get data (correctly?) from Homewizard P1 meter. I can connect directly to the Homewizard API (via the given mDNS).
Strange thing is that when I powered the controller up again yesterday it got immediately connected. After I updated to the new release v3.9.2. But when I powered the controller up today I got the same error again. I tried to reboot and reflash the firmware, but no luck.

I'm located in Belgium and got, so maybe is the P1 data structure different?

**Upload your config**
`{
  "version": "v3.9.2",
  "serialnr": 8121,
  "mode": "SMART",
  "mode_id": 3,
  "car_connected": true,
  "wifi": {
    "status": "WL_CONNECTED",
    "ssid": "VA_PRIVATE",
    "rssi": -39,
    "bssid": "80:2A:A8:9A:9F:22"
  },
  "evse": {
    "temp": 30,
    "temp_max": 65,
    "connected": true,
    "access": 1,
    "mode": 1,
    "loadbl": 0,
    "pwm": 1024,
    "custombutton": false,
    "solar_stop_timer": 0,
    "state": "Ready to Charge",
    "state_id": 0,
    "error": "Communication Error",
    "error_id": 2,
    "rfidreader": "Disabled",
    "nrofphases": 3,
    "rfid": "Not Installed"
  },
  "settings": {
    "charge_current": 0,
    "override_current": 0,
    "current_min": 6,
    "current_max": 32,
    "current_main": 32,
    "current_max_circuit": 16,
    "current_max_sum_mains": 0,
    "max_sum_mains_time": 0,
    "solar_max_import": 0,
    "solar_start_current": 4,
    "solar_stop_time": 10,
    "enable_C2": "Auto",
    "mains_meter": "HmWzrd P1",
    "starttime": 0,
    "stoptime": 0,
    "repeat": 0,
    "lcdlock": 0,
    "lock": 1,
    "cablelock": 0
  },
  "mqtt": {
    "host": "",
    "port": 1883,
    "topic_prefix": "SmartEVSE/8121",
    "username": "",
    "password_set": false,
    "status": "Disconnected"
  },
  "ocpp": {
    "mode": "Disabled",
    "backend_url": "",
    "cb_id": "",
    "auth_key": "",
    "auto_auth": "Disabled",
    "auto_auth_idtag": "",
    "status": "Disconnected"
  },
  "home_battery": {
    "current": 0,
    "last_update": 0
  },
  "ev_meter": {
    "description": "Disabled",
    "address": 12,
    "import_active_power": 0,
    "total_kwh": 0,
    "charged_kwh": 0,
    "currents": {
      "TOTAL": 0,
      "L1": 0,
      "L2": 0,
      "L3": 0
    },
    "import_active_energy": 0,
    "export_active_energy": 0
  },
  "mains_meter": {
    "import_active_energy": 0,
    "export_active_energy": 0,
    "host": "p1meter-276504.local"
  },
  "phase_currents": {
    "TOTAL": 0,
    "L1": 0,
    "L2": 0,
    "L3": 0,
    "last_data_update": 0,
    "original_data": {
      "TOTAL": 0,
      "L1": 0,
      "L2": 0,
      "L3": 0
    }
  },
  "backlight": {
    "timer": 120,
    "status": "ON"
  },
  "color": {
    "off": {
      "R": 0,
      "G": 0,
      "B": 0
    },
    "normal": {
      "R": 0,
      "G": 255,
      "B": 0
    },
    "smart": {
      "R": 0,
      "G": 255,
      "B": 0
    },
    "solar": {
      "R": 255,
      "G": 170,
      "B": 0
    },
    "custom": {
      "R": 0,
      "G": 0,
      "B": 255
    }
  }
}
`
**Upload a debug log**
`(I) (loop)(C1) STATE: A Error: 2 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer:                                                                                                                                                                                   0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A, MainsMeter.Timeout=0, EVMe                                                                                                                                                                                  ter.Timeout=64.
(I) (loop)(C1) L1: 0.0 A L2: 0.0 A L3: 0.0 A Isum: 0.0 A
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=0.0 A Imeasured=0.0 A Ma                                                                                                                                                                                  xCircuit=16 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=1.
(V) (CalcBalancedCurrent)(C1) Checkpoint 2 Isetbalanced=0.0 A, Idifference=32.0,                                                                                                                                                                                   mod=0.
(V) (CalcBalancedCurrent)(C1) Checkpoint 4 Isetbalanced=0.0 A.
(D) (CalcBalancedCurrent)(C1) Checkpoint c: Resetting SolarStopTimer, MaxSumMain                                                                                                                                                                                  sTimer, IsetBalanced=0.0A, saveActiveEVSE=0.
(V) (CalcBalancedCurrent)(C1) Checkpoint 5 Isetbalanced=0.0 A.
(homewizard_loop)(C1) homewizard_loop(): start HomeWizrd P1 reading.(getMainsFro                                                                                                                                                                                  mHomeWizardP1)(C1) getMainsFromHWP1(): invocation
(discoverHomeWizardP1)(C1) discoverHWP1(): Using cached host 'p1meter-276504.loc                                                                                                                                                                                  al'.
(getMainsFromHomeWizardP1)(C1) getMainsFromHWP1(): connect to URL http://p1meter                                                                                                                                                                                  -276504.local/api/v1/data
(getMainsFromHomeWizardP1)(C1) getMainsFromHWP1(): Error on HTTP request (httpCo                                                                                                                                                                                  de=-1), url=http://p1meter-276504.local/api/v1/data.
(I) (loop)(C1) STATE: A Error: 2 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer:                                                                                                                                                                                   0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A, MainsMeter.Timeout=0, EVMe                                                                                                                                                                                  ter.Timeout=64.
(I) (loop)(C1) L1: 0.0 A L2: 0.0 A L3: 0.0 A Isum: 0.0 A
(Timer100ms_singlerun)(C0) Unlocking Actuator.
(I) (loop)(C1) STATE: A Error: 2 StartCurrent: -4 ChargeDelay: 0 SolarStopTimer: 0 NoCurrent: 0 Imeasured: 0.0 A IsetBalanced: 0.0 A, MainsMeter.Timeout=0, EVMeter.Timeout=64.
(I) (loop)(C1) L1: 0.0 A L2: 0.0 A L3: 0.0 A Isum: 0.0 A
(V) (CalcBalancedCurrent)(C0) Checkpoint 1 Isetbalanced=0.0 A Imeasured=0.0 A MaxCircuit=16 Imeasured_EV=0.0 A, Battery Current = 0.0 A, mode=1.
(V) (CalcBalancedCurrent)(C1) Checkpoint 2 Isetbalanced=0.0 A, Idifference=32.0, mod=0.
(V) (CalcBalancedCurrent)(C1) Checkpoint 4 Isetbalanced=0.0 A.
(D) (CalcBalancedCurrent)(C1) Checkpoint c: Resetting SolarStopTimer, MaxSumMainsTimer, IsetBalanced=0.0A, saveActiveEVSE=0.
(V) (CalcBalancedCurrent)(C1) Checkpoint 5 Isetbalanced=0.0 A.
`

**P1 meter data**
`{
  "wifi_ssid": "VA_PRIVATE",
  "wifi_strength": 100,
  "smr_version": 50,
  "meter_model": "Fluvius 253769484_A",
  "unique_id": "3153414733313030303735303839",
  "active_tariff": 2,
  "total_power_import_kwh": 44868.738,
  "total_power_import_t1_kwh": 20204.339,
  "total_power_import_t2_kwh": 24664.399,
  "total_power_export_kwh": 7900.075,
  "total_power_export_t1_kwh": 5584.595,
  "total_power_export_t2_kwh": 2315.48,
  "active_power_w": -3003,
  "active_power_l1_w": -857,
  "active_power_l2_w": -1031,
  "active_power_l3_w": -1114,
  "active_voltage_l1_v": 235.9,
  "active_voltage_l2_v": 234.8,
  "active_voltage_l3_v": 235.2,
  "active_current_a": 13.16,
  "active_current_l1_a": 3.96,
  "active_current_l2_a": 4.39,
  "active_current_l3_a": 4.81,
  "active_power_average_w": 0,
  "montly_power_peak_w": 5696,
  "montly_power_peak_timestamp": 250814201500,
  "total_gas_m3": 3269.305,
  "gas_timestamp": 250831120507,
  "gas_unique_id": "37464C4F32313139313530303434",
  "external": [
    {
      "unique_id": "37464C4F32313139313530303434",
      "type": "gas_meter",
      "timestamp": 250831120507,
      "value": 3269.305,
      "unit": "m3"
    }
  ]
}`

**To Reproduce**
Steps to reproduce the behavior:
Smartevse is configured in smart mode. Just starting up the controller.

**Expected behavior**
Receive the P1 meter data.

dingo35, 2025-08-31:
So on which version were you first, when it worked?

dingo35, 2025-08-31:
Could you try current stable v3.8.5?

dingo35, 2025-08-31:
And: your SmartEVSE is on the same LAN as your HomeWizard? No VLAN's or otherwise segmented network?

joyvanacker, 2025-08-31:
On the stable V3.8.5 same problem, same network, No seperate VLAN's or anything else.

BUT: I forced the SmartEVSE on the same AP as the P1 meter and now it works. Very strange because I have no problems with other integrations and/or connections whatsoever.

Thanks to thinker about the problem with me anyway! Maybe helpful in a way for other users in the future :-)

dingo35, 2025-08-31:
Might be; with a response time of 2.7s you definitely have a network issue; either your different AP's are not fully bridged (which is necessary for mDNS), your wifi is on a busy channel, your AP's are disturbing eachother, you might have some network loop.

Better analyze whats going on on your LAN with wireshark....