dingo35/SmartEVSE-3.5 #275: MQTT stays "Disconnected" on HiveMQ Cloud (v3.9.2, ca_verify problem?)

Dnanilem2, 2025-09-07:
Hello,
**SmartEVSE version**: dingo35 v3.9.2
**Serialnr**: 8173
**Mode**: NORMAL

**Issue**:
I’m trying to connect SmartEVSE to HiveMQ Cloud with:
- host: 0ce2592bd83d46bb9fc072af7808ae94.s1.eu.hivemq.cloud
- port: 8883
- username: Dnanilem
- password: set
- topic_prefix: SmartEVSE1947

Wi-Fi is connected (WL_CONNECTED), but `mqtt.status` remains `"Disconnected"`.  
With the same parameters, my Shelly Pro 3EM connects fine.

**What I tried**:
- checked username/password on HiveMQ Cloud
- time/date is correct
- tried changing topic_prefix and client_id
- suspect `ca_verify` or TLS root CA validation issue

**Question**:
This configuration does not require a certificate and works perfectly with my Shelly meter.
My goal is to establish a direct connection to the HiveMQ server that I already use for other applications.
Could you please provide some guidance on how to solve this problem? It would also be great if the configuration for this meter could be supported in your software.
Thank you in advance.

- Is there a way to disable `ca_verify` or upload DigiCert CA?
- Any known limitation with HiveMQ Cloud TLS in v3.9.2?

dingo35, 2025-09-07:
Only plain MQTT is supported, no tls.

mstegen, 2025-11-28:
I've just added TLS support.

Could you test the attached firmware?

I've succesfully tested it on HiveMQ
You can paste in the CA cert. HiveMQ uses letsencrypt, so you can use this cert:
https://letsencrypt.org/certs/isrgrootx1.pem

[firmware.zip](https://github.com/user-attachments/files/23825146/firmware.zip)

mstegen, 2025-12-09:
closed due to inactivity

closed 2025-12-09 by mstegen