dingo35/SmartEVSE-3.5 #280: eGolf 2018 not charging with releases since commit 5f391d1

Lordhu, 2025-09-16:
Background, household with two EV's: 1x eGolf (2018), 1x Tesla, 2x SmartEVSE V3

Recently decided to upgrade to the latest release due to all the cool new features.

Problem:
The Tesla works fine with the newest releases (tested v3.9.2 and also v3.8.5) on both SmartEVSE's
Unfortunately the eGolf gets into an endless loop of charging / ready / charging / ready / etc. 
There is a few seconds between each state change. The Tesla works perfectly fine on the same SmartEVSE/charger.

Managed to narrow the problem down:
Release v3.6.10 and older work perfectly for the egolf
Release v3.7.1 and newer do not work the egolf

By looking through the commits between the two versions I suspected that commit 5f391d1 might cause the symptoms.
I compiled a recent version with commit 5f391d1 disabled: the eGolf was charging fine again.

Commit 5f391d1 states that the ESVE should switch to state b if the pilot signal is not indentified as PILOT_6v  
By compiling some more versions I managed to find out that the pilot signal of the eGolf is measured by the ESVE as PILOT_NOK

With a multimeter I measured the voltage between CP and PE: the multimeters shows around 6v while charging. I think this does mean there is no short circuit between CP and PE (if shorted than there should be no voltage between CP and PE)

Tested further and managed to find out that the 6V Pilot signal of the eGolf is outside the range accepted by the SmartEVSE as soon as the current to the car starts flowing. The PILOT_6V signal is supposed to be between Min >= 2400) & (Max < 2735) (see code section below)
The Min value of the eGolf PILOT_6V signal seems to be around 2345-2355 while charging at 16a (determined this by compiling with different values until the charging was stable again). This means the Pilot signal is not indentified as PILOT_6V (as the min value is too low for PILOT_6V) but also not as PILOT_3V (as the max value is too high for PILOT_3V). This means the Pilot state will automatically be PILOT_NOK, what stops the charging since commit 5f391d1 is introduced.

Probably the egolf doesn't completely follow the rules here (piot signal is probably fluctuating quite a bit); but if I understand correctly commit 5f391d1 is made to detect a short circuit between CP and PE, and this is not the case here.

Can we maybe make the CP to PE short circuit detection be made a littlebit less sensitive?

Maybe this can be done by creating a seperate variable for CP to EP short circuit, and use only that variable to switch from state C to state B. Or an alternative solution can be to create a bit of overlap between the PILOT_6V Min and PILOT_3V Max values in the Pilot signal catogories (I have used this solution for a few days now: changed the MIN value for PILOT_6V to 2340 and this seems to be working so far)

Related code:
----------------------

Pilot signal measurement (main.cpp):

    // calculate Min/Max of last 25 CP measurements
    for (n=0 ; n<25 ;n++) {
        sample = ADCsamples[n];
        voltage = esp_adc_cal_raw_to_voltage( sample, adc_chars_CP);        // convert adc reading to voltage
        if (voltage < Min) Min = voltage;                                   // store lowest value
        if (voltage > Max) Max = voltage;                                   // store highest value
    }
    //_LOG_A("min:%u max:%u\n",Min ,Max);

    // test Min/Max against fixed levels
    if (Min >= 3055 ) return PILOT_12V;                                     // Pilot at 12V (min 11.0V)
    if ((Min >= 2735) && (Max < 3055)) return PILOT_9V;                     // Pilot at 9V
    if ((Min >= 2400) && (Max < 2735)) return PILOT_6V;                     // Pilot at 6V
    if ((Min >= 2000) && (Max < 2400)) return PILOT_3V;                     // Pilot at 3V
    if ((Min > 100) && (Max < 300)) return PILOT_DIODE;                     // Diode Check OK
    return PILOT_NOK;                                                       // Pilot NOT ok
}
#endif

--------------------

Code of commit 5f391d1: (main.cpp) switch to state B if PILOT signal is not indentified as PILOT_6V

        } else if (pilot != PILOT_6V) {                                     // Pilot level at anything else is an error
            if (++StateTimer > 50) {                                        // make sure it's not a glitch, by delaying by 500mS (re-using StateTimer here)
                StateTimer = 0;                                             // Reset StateTimer for use in State B
                setState(STATE_B);
                DiodeCheck = 0;
#ifdef SMARTEVSE_VERSION //not on CH32
                GLCD_init();                                                // Re-init LCD (200ms delay); necessary because switching contactors can cause LCD to mess up
#endif

mstegen, 2025-10-01:
Thanks for your analysis. 

I will see if i can change the code so that it still detects a short between CP and PE, but doesn't stop charging on your eGolf.

Lordhu, 2025-10-04:
> Please test the attached firmware. I introduced a new define PILOT_SHORT with a range from 1600-2000. This still detects shorts between CP and PE, but should fix your issue.
> 
> [firmware.zip](https://github.com/user-attachments/files/22664557/firmware.zip)

Thank you! I have uploaded the firmware and first impressions are very good! (battery was alsmost full so I could not test for very long) Will check the charging the next few days and report back

Lordhu, 2025-10-04:
> Just to inform you, my e-Golf (2018, 36Kwh model) is charging perfectly with the v3.9.2 version. My problem was solved and turned out to be an installation issue.

That is strange. One difference might be that my eGolf is probably on the original software (as it was sold and serviced by a non-VW dealer its entire life).

mstegen, 2025-10-07:
Perfect! As you might have noticed, i just pushed the required changes to this repo https://github.com/dingo35/SmartEVSE-3.5/commit/fcf19c20c7332cd0b0ccedf255617c7bfb88afd7